CNCKitchen/ExtrusionSystemBenchmark #2: M109 Wait for Extruder missing

TheNinsa, 2022-04-03:
In the Start GCODE, while not necessary in some cases, the M109 is missing.

While trying the test again with different parameters, the Bed was roughly at temperature, while the extruder was not. The missing M109 caused the printer to not heat up properly and the print to fail.

TheNinsa, 2022-04-03:
FYI: I am using an Ender 3, that is running on Klipper. I had to increase the Maximum Crosssection as well as the Maximum Extrude Only Move to 2.5 and 250mm respectively.

TheNinsa, 2022-04-03:
While Testing, it seems that the generated Gcode is also not keeping the Temperature.
[image]

TheNinsa, 2022-04-03:
I am updating this as I continue to test.
First edit: I thought the M109 was missing completely, but looking at the gcode, it is set at the beginning of each temperature setting. Weirdly enough, it doesnt keep the temperature.

Second edit: Klipper apparently does not support the R Prefix on M109, I changed it for S, and am running the test again

TheNinsa, 2022-04-03:
I can confirm: the " Malconfigured" M109 is the "Issue".

This is purely to the fact that I am not running Marlin, but Klipper. And Klipper does not support M109 R[HotendTemp]

As additional feedback: It would be lovely to be able to add ones own StartGcode from the GetGo. I store most of my Starting Parameters and Logic within a GcodeMacro in Klipper.

appositeit, 2022-04-03:
Klipper doesn't support the R parameter for the M109 command, only S. Search replace s/M109 R/M109 S/ is quick fix for the generated gcode.

Marlin has different behaviours for the R and S parameters:
[R<temp>] | Target temperature (wait for cooling or heating).
[S<temp>] | Target temperature (wait only when heating). Also AUTOTEMP: The min auto-temperature.

Klipper implements S behaviour only and it is equivalent to R on Marlin:
Set extruder temperature and wait: M109 [T<index>] S<temperature>
Note: M109 always waits for temperature to settle at requested value

So it needs to be M109 R on Marlin and M109 S on Klipper.

TheNinsa, 2022-04-03:
Ah that makes more sense, Thank you :D

closed 2022-04-03 by TheNinsa