CNCKitchen/ExtrusionSystemBenchmark #10: On Klipper "M109 Rxxx" will turn off extruder heater

matthew-humphrey, 2022-04-06:
I generated some G-Code with the spreadsheet, and on the first blob noticed my extruder clicking. This turns out to be because the temperature had been reset to zero. I looked into it, and it seems that on Klipper, a command like M109 Rxxx will set teh extruder temperature to zero regardless of the value of xxx. The Klipper docs say that M109 Sxxx will always wait for the temp to stabilize at the requested value. 

So perhaps add an option to generate M109 S instead of M109 R for those of us using Klipper.

Chimorinxx, 2022-04-07:
can't you make a macro in the Config to replace M109 Rxxx with M109 Sxxx? i mean in the Config of Klipper. i don't know how to change the macro of this Spreadsheet so i would likely do that if it worked. Haven't tested it yet.

matthew-humphrey, 2022-04-08:
I also just changed the Excel macro locally. It does work.

blalor, 2022-04-17:
The difference between Klipper and Marlin can be worked around by ensuring the temperature steps *up*; the spreadsheet defaults to a -20K step. If instead the temperature always _increases_ `M109 Sxxx` can be used consistently among the supported firmwares.

matthew-humphrey, 2022-04-17:
Much simpler and easier to just modify the macro. I was just sharing in case the author wanted to address the issue.

matthew-humphrey, 2022-06-04:
> Hi matthew. Im willing to run the same test on my printer. I also ahve klipper. The only thing you changed is the R to S, correct? nothing else?

Hi Neiizo. Yes, that's correct.