CNCKitchen/ExtrusionSystemBenchmark #15: Option for direction

Avatarsia, 2022-04-09:
i have a cooling fan duct in from of the printer thats very close to the heat bed. 
so if you make the first test is will squeeze the first test result during the 2nd test is made.
this could be avoid with a start at the rear and first do all test in X direction

photodude, 2022-04-10:
@Avatarsia 
There is a direction option in the existing code for reversing Y direction.  I believe it's on line 22 of the first excel tab. setting the direction field to `-1` will set back to front printing (reverse Y.)  Are you asking for something similar but for X-direction ordering?

I also have issues with printing this due to the gantry carriage hitting the prints. seems I'm limited to a height of about 5mm or or really far flow spacing I hit the prints with my dual extruder.

Avatarsia, 2022-04-11:
i need to check that option. 
but in general a pettern choise would be a nice idea