CNCKitchen/ExtrusionSystemBenchmark #19: klipper issue: move exceeds maximum extrusion

Neiizo, 2022-06-05:
Hi. I'm using klipper and was really looking forward to using this tool. However, the code doesn't even start. After heating up, it gets into position and I get the message:
Move exceeds maximum extrusion (1.924mm^2 vs 0.640mm^2)

I'm not sure how to fix this. Can anybody help ?

PaFcio, 2022-07-08:
You need to add following to [extruder] entity in config:

max_extrude_cross_section: #I used 5 here, worked
max_extrude_only_accel: #I used 5 here, worked
max_extrude_only_distance: #here 200 recommended as that is what is extruded afaik

m0n5t3r, 2024-02-23:
what you actually need is `max_extrude_only_distance` and `max_extrude_cross_section`; setting `max_extrude_only_accel` to 5 had me scratching my head for a day trying to figure out why klipper appeared to stutter randomly (retraction / unretraction is an extrude-only move, and setting an acceleration so low causes even a 0.5mm retract / unretract to take ~ 2 seconds)